VirtualBrainLab/Pinpoint #773: Unable to add account/save experiment

adeetiaggarwal, 2025-04-13:
**Describe the bug**
I am unable to find the accounts or save experiments tab in pinpoint to save my experimental coordinates. This is both in the web version and in the version that I have on my PC through Stream which I just updated. 

[image]

[image]

kjy5, 2025-04-13:
Hi,
V2.0.0 removed the accounts system as we change to a new model. If you need the accounts, please use [v1](https://github.com/VirtualBrainLab/Pinpoint/releases/tag/v1.1.7) which you can find  by downloading v1.1.7 or changing your [Steam branch to v1](https://partner.steamgames.com/doc/store/application/branches).

closed 2025-04-14 by kjy5

dbirman, 2025-04-14:
@adeetiaggarwal just adding a note here -- we're going to transition to a different system for v2 where you can export/save the pinpoint scene directly as a file or store it on a server that we'll maintain. We'll eventually shut down the old accounts server because the it's too expensive to maintain, but you can keep using until we have the new system running.