VirtualBrainLab/Pinpoint #779: Cannot move Sensapex uMp-4 manipulator via Pinpoint

AmirKhalili12, 2025-05-02:
Hello, 

I’m currently setting up our Sensapex uMp-4 manipulators to be controlled via PC using Pinpoint and the Ephys Link interface. The system is correctly detecting both manipulators, and position echoing works — meaning when I move the manipulators manually (via the TSC or rotary controller), the movement is reflected accurately in Pinpoint.
However, I’m unable to send movement commands from the PC to the manipulators. The Ephys Link console displays the following error:
ValueError: Device #1 is configured for 3 axes, but target position has 4 elements.

From what I understand, this is likely due to Pinpoint sending a 4-element command but detecting uMp-4 as a 3-axis manipulator.
I’ve looked through the Pinpoint interface but  I have not found a way to disable pitch/yaw/roll or to limit movement commands to 3 axes or to detect uMp-4 as a 4-axis manipulator. I would greatly appreciate your assistance regarding this matter. If you need any additional informationI would be happy to provide it. A picture of the error mentioned above is attached here as well. 

[image]

kjy5, 2025-05-02:
Hi,
That's an interesting error. It looks like Pinpoint believes it has connected to a uMp-3 manipulator. Can you send a screenshot of the connection window in Pinpoint? Also, what version of Pinpoint are you using? v2.0.0 should require you to select "Sensapex uMp-4" during setup to connect to uMp-4 manipulators. 

Thanks!

kjy5, 2025-05-07:
@AmirKhalili12 Were you able to resolve this issue on your own?

kjy5, 2025-05-11:
Actually, looking closer at the error message, it appears your manipulator 1 is a uMp-3, and (presumably) your manipulator 2 is your uMp-4. Ephys Link currently does not support mixing these two manipulators in a rig yet. If this is a pressing issue, you can open a _feature request_ for that.

For now, if you plan on using uMp-4, please remove the uMp-3 manipulator from the rig. Future versions of Ephys Link will automatically select uMp-3 or uMp-4 support based on the **first** manipulator it sees.

closed 2025-05-11 by kjy5

AmirKhalili12, 2025-05-19:
@kjy5 
Hello,
Sorry for the delay in my response.
We're currently using Pinpoint v2.0.0 from Steam. I'm not entirely sure what the "connection window" refers to, but I've attached a screenshot of the Ephys Link tab in Pinpoint for reference:

[image]

You were correct in mentioning that we might be using a uMp-3 manipulator. However, Pinpoint is only able to connect to the manipulators using the uMp-4 server connection which is why we were led to believe we were using a uMp-4 manipulator. Here is the error when trying to connect as a uMp-3.

[image]

We’re stuck in a situation where:
The manipulator echoes position fine in Pinpoint.
Movement commands from Pinpoint to the manipulator fail due to axis mismatch.
The uMp-4 setting is the only one that successfully connects, even though the system is a uMp-3.

Any guidance you can offer for resolving this mismatch would be greatly appreciated.

Thank you again!

kjy5, 2025-05-19:
Thanks for the update! Looks like for some reason you're not able to launch the uMp-3 connection. I'd suggest trying the beta version of Pinpoint that will automatically detect which manipulator you are using. [Follow these instructions](https://developer.valvesoftware.com/wiki/Betas#For_the_Beta_version_of_the_game:~:text=to%20do%20so.-,For%20the%20Beta%20version%20of%20the%20game,You%20may%20only%20select%20one%20beta%20program%20at%20a%20time.,-Beta%20removal) to switch your Steam branch to "ephys-link-betas". This will get you going for now. Once this becomes the main release you can switch out of the betas.

AmirKhalili12, 2025-05-19:
@kjy5,

That worked perfectly, thank you so much for the quick help!

Switching to the ephys-link-betas branch resolved the issue, and the manipulators are now responding correctly to movement commands from Pinpoint. We’ll keep using this beta version for now and switch back once it’s merged into the main release.

Really appreciate your support!

kjy5, 2025-05-19:
Sweet! Great to hear the issue was resolved. Feel free to reach out again if anything goes wrong!