VirtualBrainLab/Pinpoint #780: Home positions for different atlases

20180220, 2025-05-07:
Hello,

Actually we are trying different atlases. We noticed that the home position for mouse CCF(25um) is right above bregma. However, the home position for other atlases tend to have it somewhere else. We are wondering if there is any way to change the home positions? We are sorry but we could not figure it out. 

Best Regards,

kjy5, 2025-05-07:
Hi,

It looks like the reference coordinate got reset. Go into the menu using the <kbd>h</kbd> key, then select "Atlas", and then set the reference coordinate to Bregma by pressing the Bregma button or enter a custom AP/ML/DV coordinate:

[image]

Let me know if this fixes your problem!

20180220, 2025-05-09:
Yes, we were able to reset it! Thank you very much!

Best Regards,

closed 2025-05-09 by kjy5